pcdshub/happi #100: Force beamline to be all caps

ZLLentz, 2020-03-03:
## Expected Behavior

Force names like XPP, LFE, MFX

## Current Behavior

Allows names like xpp, lfe, mfx

## Context

Naming consistency is good

## Your Environment

ZLLentz, 2020-03-03:
This should come along with a case-insensitive search mode for case-insensitive fields so modules like lucid don't have to do special processing on specific fields.

See https://github.com/pcdshub/lucid/issues/46#issuecomment-594068912

cristinasewell, 2020-08-27:
Can I assume that the `beamline` only consists of letters from [A-Z]? or is it possible that we have other characters in there too?

ZLLentz, 2020-08-27:
At SLAC/LCLS, only letters is a good assumption. All of our beamlines are things like XPP, XCS, LFE, CXI.

Since the initial posting of this issue, we've migrated our beamline schema to https://github.com/pcdshub/pcdsdevices/blob/master/pcdsdevices/happi/containers.py
Since Hugo refactored this library to allow us to define containers in other repositories.

The idea is that `happi` itself should not make any particular assumption as to which facility it is being used in, but a facility can extend the schema for its own purposes. That is to say, `happi` itself shouldn't even know what a `beamline` is.

I think I will transfer this issue over the pcdsdevices repo and update the comment