pcdshub/happi #106: Discussion about Device and Mandatory Fields

hhslepicka, 2020-03-10:
## Expected Behavior
Since other labs are looking into Happi, should we make Device be completely flexible with additional metadata and create the PCDSDevice or some other name that will have mandatory fields such as `functional_group` and `location_group` ?

## Current Behavior
Device requires `functional_group` and `location_group` along with some other information that is a bit LCLS specific.

## Possible Solution
Create a new type of container that is the PCDS basic Device and make Device very relaxed.

hhslepicka, 2020-03-10:
From SLACK talk:
- Zach says: "My answer is yes"
- Ken says: 👍

ZLLentz, 2020-03-10:
I think we should use this opportunity to change our semantics too. These should never have been called `Device` because of the conflict with `ophyd`. I think `HappiItem` should be the super basic entry, then if you want an ophyd device entry it should be `DeviceItem`, and then more specific should be `SlitsItem`, etc. if we want these.

I would also advocate for pulling all SLAC-specific assumptions out of `happi` and into e.g. `happi_lcls`

ZLLentz, 2020-03-11:
There would also need to be a way to register your site-specific classes with `happi`

klauer, 2020-03-11:
Not that I'm against it necessarily, I think pulling things out into a separate package for PCDS/LCLS is probably slightly overkill at this point.

A minimum viable thing to make this useful for others and still keep PCDS stuff our priority would be refactoring as @hhslepicka suggests, I think.

hhslepicka, 2020-03-16:
Discussion from #112:

> The container discovery will need most of the refactor.(https://github.com/pcdshub/happi/blob/master/happi/client.py#L44-L52)
> 
> Here is one idea:
> Make containers be converted from a python file to a namespace package (https://packaging.python.org/guides/packaging-namespace-packages/#creating-a-namespace-package) with a registry inside of it.
> Facilities that wants to add new containers can create a package which will share this namespace and be discovered by Happi when running.
> 
> Another idea (by @klauer):
> Investigate the usage of entry_points so that we don't need to enforce a package architecture into projects that want to extend the containers for Happi.

PR #126 (2020-04-16): ENH: Containers via entrypoints

## Description
As of now in order to add a new container we need to modify Happi source code and add it at the `happi.containers` module so it is found by the client.
Via discussion at #106 and many talks with @ZLLentz and @klauer we decided to experiment with the usage of entry_points.
The LCLS specific containers and item were moved to `pcdsdevices` under the following PR: https://github.com/pcdshub/pcdsdevices/pull/402

## Motivation and Context

Merging this will contribute to close #106 

## How Has This Been Tested?

Tested locally

## Where Has This Been [... cut]
closed 2020-04-16 by klauer