pcdshub/happi #124: Deprecate "device" verbiage for classes from Happi

ZLLentz, 2020-03-20:
Use `item` instead of `device` to clarify which things are actual functioning devices and which things are descriptions of functioning devices

Todo:
- [ ] Deprecate `Device` class, replace with `OphydItem` and `LCLSItem`
- [ ] Deprecate `device` module, replace with `item`
- [ ] Deprecate methods with `device` in the name except where they return a user's instance, replace with `item`
- [ ] Adjust comments, docstrings, and internal code to clarify types
- [ ] Clean up the docs in general

Related:
- [ ] Move `LCLSItem` and LCLS's containers into `pcdsdevices`

ZLLentz, 2020-03-20:
Started in #122

klauer, 2020-04-06:
- [x] `SearchResult.item`, `Client.all_items` and deprecate `SearchResult.device`, `Client.all_devices`

ZLLentz, 2021-01-27:
I'm currently getting a headache trying to read through the Client docstrings- it uses device, instantiated device, item, etc. seemingly interchangeably.

klauer, 2021-01-27:
It's slowly improving but still bad, for sure. Wish we had gotten rid of 'device' longer ago.

closed 2022-06-22 by klauer