pcdshub/happi #125: Rework of args, kwargs entries

ZLLentz, 2020-03-20:
## Expected Behavior

By default, `happi` should inspect the `device_class`'s signature and automatically set matching keys from the database. The `args` and `kwargs` keys should only be used for devices with `*args` and `**kwargs` behaviors. This way there is no need for a custom container for each device class, except to add necessary metadata or special parameters in a clean way.

e.g. a config like:
```
HappiItem
device_class: EpicsMotor
prefix: MY:PV:PREFIX
name: my_name
```
Should be enough to make it work. Then you could add in:
```
kwargs: {settle_time=1.0, timeout=10}
```
Without needed to keep the original structure in mind, e.g. right now you'd need to:
```
kwargs: {name={{name}}, settle_time=1.0, timeout=10}
```

## Current Behavior

`args` and `kwargs` are fields that are parsed to decide which keys to pass into the device class's constructor. The defaults are currently  `args=['{{prefix}}']` and `kwargs={'name'='{{name}}'}` for `OphydItem`, and you would need to expand these to insert additional args and kwargs. `HappiItem` currently cannot be used as a container at all without inputting unique values for `args` and `kwargs`, even if they echo metadata entries.

## Possible Solution

`inspect.getfullargspec` at device instantiation, then add additional `*args` and `**kwargs` in as directed.

ZLLentz, 2020-03-20:
Just an idea, there are probably other ways to approach fixing the unwieldiness of args/kwargs arguments in `happi`

klauer, 2020-03-23:
This is great. @hhslepicka or @ZLLentz, what's your interest level in tackling this?

ZLLentz, 2020-03-23:
@klauer I'm definitely interested, but I don't have time until I finish all my tasks that are either "T1" on the trello board or blocking other engineers.

hhslepicka, 2020-03-23:
I can take a stab at it after the other PR for the entrypoints. Would that be ok?

ZLLentz, 2020-03-23:
I'd be happi for anyone to tackle this one because