pcdshub/happi #131: Cannot change device name field using json backend

jsheppard95, 2020-04-16:
## Expected Behavior
Instantiating a `happi.Client` in an IPython terminal, one may try to edit a device name as follows:
```
import happi
client = happi.Client(path='/reg/g/pcds/pyps/apps/hutch-python/device_config/db.json')
dev = client.find_device(name='mr1l0')
dev.name = 'mr1l0_homs'
dev.save()
```
## Current Behavior
Saving the device as shown above (last line) results in the following error:
```
In [14]: dev.save()
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
/reg/g/pcds/pyps/apps/dev/pythonpath/happi/backends/json_db.py in save(self, _id, post, insert)
    280                 try:
--> 281                     db[_id].update(post)
    282                 except KeyError:

KeyError: 'mr1l0_homs'

During handling of the above exception, another exception occurred:

SearchError                               Traceback (most recent call last)
<ipython-input-14-f5fa237e2afe> in <module>
----> 1 dev.save()

/reg/g/pcds/pyps/apps/dev/pythonpath/happi/client.py in <lambda>()
    290
    291         # Add the save method to the device
--> 292         device.save = lambda: self._store(device, insert=False)
    293         return device
    294

/reg/g/pcds/pyps/apps/dev/pythonpath/happi/client.py in _store(self, device, insert)
    594         # Store information
    595         logger.info('Adding / Modifying information for %s ...', _id)
--> 596         self.backend.save(_id, post, insert=insert)
    597         return _id
    598

/reg/g/pcds/pyps/apps/dev/pythonpath/happi/backends/json_db.py in save(self, _id, post, insert)
    281                     db[_id].update(post)
    282                 except KeyError:
--> 283                     raise SearchError("No device found {}".format(_id))
    284
    285     def delete(self, _id):

SearchError: No device found mr1l0_homs
```
This works for all other happi fields, but not for `name` likely due to a connection with the `_id` field.
Instead, to change the device name one must simply delete the happi entry and re-add it back in with the new name.

## Possible Solution
It seems like the `SearchError` generated by the `save()` function call is due to searching the `new_name` before the happi entry is updated. It may be as simple as having a variable `old_name` that can be used for searching the database.

## Your Environment
Using `dev_conda` environment:
`/reg/g/pcds/pyps/conda/dev_conda`

ZLLentz, 2020-04-16:
Another workaround is to make a new `happi` entry with the same fields but a different name using the command-line `--clone` argument, and then deleting the old entry.

cristinasewell, 2020-08-13:
Because the name is tied to the ID of the entry, changing the name itself is possible but i don't think it is desirable, because it brings the possibility of being able to have multiple entries with the same name which can be a disaster, but in this case we would have to change the `_id` as well -  in other words we might have to remove the entry first and then add it back with the new name - something that `--clone` would do?

How do I clone? Can anybody help with with cloning please - i am trying to clone and getting an error (`AttributeError: 'HappiRegistry' object has no attribute 'entry_from_class'`), so I think I am doing something wrong, but can't tell what?

ZLLentz, 2020-08-13:
Or maybe cloning is bugged... I get the same error.

PR #157 (2020-08-18): MNT: fix entry renaming

## Description

`client.py`'s **`_store`** method has been modified to store the old_name to be able to find the entry in case the user is trying to change its name. An extra check has been added to see if the user is trying to change the name of an existing entry, if so, a new entry is created with a new name and the older entry gets deleted after that. 

## Motivation and Context

These changes allow the users to change the 'name' of an entry. 
Because the 'name' is tied to the '_id' of an entry (as of now, this might change in the future), it is not practicable to c [... cut]
closed 2020-08-18 by ZLLentz