pcdshub/happi #141: CLI glob/regex search

ZLLentz, 2020-06-02:
## Expected Behavior

One of these should help me find the kmono entry:
`happi search kmono`
`happi search *kmono*`
`happi search .*kmono.*`

## Current Behavior

To find the kmono entry, you must precisely request:
`happi search sp1l0_kmono`

## Possible Solution

Use search_regex instead of search
Add a search_glob and use that

cristinasewell, 2020-07-30:
A simple solution would be to modify the find_regex function to use `re.findall` instead of `match` - and use the `search_regex` when searching instead of the normal `search`  like Zachary suggested ( or we can add a new search_glob - the other suggestion, and do the same ) 
This will also solve the problem with case sensitivity when searching, but I am wondering if this could introduce other problems that i can't think of right now. (i know it will break some tests at least) 
@klauer is there a specific reason we want to use regex.match when searching with `find_regex` - because I am still learning about this I am not aware of all the scenarios, what is your opinion on this? (i'm asking you cause i saw you worked on these functions :) )

klauer, 2020-07-30:
I think @ZLLentz is saying that one of those should provide a result, not that all of them need to.

The support is in there now and uses IGNORECASE for flags by default:
https://github.com/pcdshub/happi/blob/bf69207d73113271afb1c2e33bdb3cb7e3af24ed/happi/client.py#L459-L486

From an API standpoint, `match` is appropriate in my opinion as it can be used for either searching or exact matching.
`search('x') -> match('^.*x.*$')`

You can also convert glob strings, making this something the backend doesn't really have to deal with, I think. (see https://docs.python.org/3.7/library/fnmatch.html#fnmatch.translate - there are examples in typhos where I've used this, too, if it helps)

cristinasewell, 2020-08-03:
Cool, thanks for the suggestions @klauer . I like `fnmatch.translate` - works very well with everything except the `z` value (numeric values) in this case - i need to look more into it, i might need to add some additional tweaking when working with numbers :) 
Thank you!

cristinasewell, 2020-08-10:
@ZLLentz do we also want this glob/regex search to be applied to the `z` values? something like this:  `happi search z=6.*`  or do we want to be using the normal `search` method for `z` values?

ZLLentz, 2020-08-10:
You bring up a good point, glob/regex isn't so intuitive when it comes to numeric fields.

I think for now we shouldn't treat them specially, since being able to do a regex here is much more powerful than only being able to target a single position on the beamline. But we need to answer a design question about the most intuitive way to have the user search for a range when we have a floating point field like this.

cristinasewell, 2020-08-10:
Yeah that sounds good - thank you. 

Some ideas:

1. `happi search z=1035.8` -> should match the device exactly at 1035.8

and at least one of these: 

2. `happi search z=1035.*`-> should match all the devices at positions that start with 1035. ? basically from 1035-1036 (not including 1036)
3. `happi search z=1035.0-1035.8` -> or we can simply define a range here
4. `happi search z=[1035.0-1035.8]` -> same here
5. `happi serach z=... some better ideas here ` :) 

@klauer  what do you think?

ZLLentz, 2020-08-10:
The range selectors have built-ins in the API for searching for them, so those are the most preferable from a developer standpoint. In my head, the following are basically the same for a user:
`z=1035.0-1035.8`
`z=[1035.0-1035.8]`
`z=[1035.0,1035.8]`
`z=(1035.0,1035.8)`

hhslepicka, 2020-08-10:
Sorry if this is a stupid question but can we ever have a negative `z` value?

cristinasewell, 2020-08-10:
Thanks for asking that Hugo, that's a great question, I am curious to know the answer to that too, since I would have to consider it in this Issue

ZLLentz, 2020-08-10:
It is not a stupid question.
No device in LCLS will ever have to deal with negative z.

But happi should handle this elegantly regardless of what the float field represents and should be aware that negative values are possible. I assume this rules out elegant handling for the `z=1035.0-1035.8` and `z=[1035.0-1035.8]` formulations.

cristinasewell, 2020-08-10:
One more question here, I am not sure if i will need this but I'd like to know: 

Could we, for example,  have a device at `1035.25` and a different device at `1035.28` - or in general, is there a certain accuracy that we could have for positions?

ZLLentz, 2020-08-10:
The database right now is extremely inconsistent. Sometimes the position is estimated, and sometimes the position is read directly from an engineering drawing with precise measurements (and in some rare cases, someone went into the tunnel with measuring tape...)

I think the trailing zeroes in the database are an artifact of the saving: the person almost certainly just entered `89` when adding that device, though I know of at least one case where the engineering drawing specified four zeroes...

I have to suspect that, in general, we're just casting this input to a float and letting the json library handle the details of the saving.

cristinasewell, 2020-08-10:
Ohh cool, thank you 👍

PR #155 (2020-08-18): Cli search regex

## Description
- ` happi search` command line has been modified to use `search_regex` and `search_range` (for z values) instead of the previous `search` function. 

## Motivation and Context
- this change will allow the user to search using the asterisk symbol `*` that matches 0 or more occurrences.
    ex: `*kmono*` will match any occurances that have `kmono` in them
- this also allows the user to search for a range of `z` values by typing a valid range separated by comma like so:
  - z=0,89
**Note:** there is no space between the values

Intended to close #141 
Will possibl [... cut]
closed 2020-08-18 by ZLLentz