pcdshub/happi #142: CLI happi audit

ZLLentz, 2020-06-02:
## Expected Behavior

Command to audit the happi database for entries that are not loadable, and help diagnose why. For bonus points, command that helps us migrate these bad entries and make them good.

## Current Behavior

No easy way to debug malformed entries (perhaps from old versions of happi)

## Context

There are devices in the current deployed db at LCLS that are broken and I do not know why

hhslepicka, 2020-06-02:
Can it be due to the changes we introduced with the moving from happi to pcdsdevices for the containers?

ZLLentz, 2020-06-03:
This is possible, it's also possible that it's from schema changes that I made.

cristinasewell, 2020-08-28:
As of now I can insert an OphydItem or HappiItem through the client with extra information, like this: 

```
client = happi.Client(path='../example/db.json')
device_info = {'name': 'alias2',
            'z': 400,
            '_id': 'alias2',
            'prefix': 'BASE:PV',
            'beamline': 'LCLS',
            'type': 'OphydItem',
            'device_class': 'types.SimpleNamespace',
            'args': list(),
            'kwargs': {'hi': 'oh hello'},
            'location_group': 'LOC',
            'functional_group': 'FUNC',
            }
#device = OphydItem(**device_info)
device = HappiItem(**device_info)
client.add_device(device)
```

It adds it fine to the database, and it loads it as well, we can also edit certain fields, but not the ones that the `OphydItem` or `HappiItem` don't have (like `beamline` for ex.) 

This makes it a bit challenging to audit - because it loads the entry fine, yet I think it is a bad entry right?. 
Theoretically, should this type of entry not be allowed to be loaded or do we not want that?

I am wondering if for this audit command we could use something like `jsonschema` and a schema we can check against depending on the item..... I am not sure how complicated it would get though... What are your thoughts on this?

ZLLentz, 2020-08-28:
I'm not sure if this should be allowed or not. On the one hand, it may be useful to attach extra metadata to your happi entries for specific use cases. On the other hand, this makes it difficult to understand the database from a schemas perspective if any entry can have arbitrary extra keys.

My gut says that the audit shouldn't fail for these right now, and we should continue allowing them, but surely this deserves a warning, something like: "Warning, happi item alias2 has the following extra keys beyond the entry info definition for OphydItem: <keys>. Consider creating a custom happi container to validate these fields."

ZLLentz, 2020-08-28:
RE: jsonschema, I saw that library recently for the first time and thought it looked really useful, but I don't have enough experience to know how easy or complicated it would be to apply to this use case.

cristinasewell, 2020-08-29:
`jsonschema` is pretty cool, i am just not sure if i'll be limited with it somehow. What i like about it is that depending on how well the schema is defined, i can get very specific error messages for any fields in the entry. 
For example I created a very basic schema for `OphydItem`, and I specified in the schema that I don't want the OphydItems to have additional properties other than the ones I specifically defined (I know this is not what we want, i'm just given here an example), then if it finds one Entry with multiple properties it will give me this message (which is pretty good and specific): 

```
Additional properties are not allowed ('functional_group', 'z', 'location_group', 'beamline' were unexpected)

Failed validating 'additionalProperties' in schema:
    {'additionalProperties': False,
     'properties': {'_id': {'type': 'string'},
                    'active': {'type': 'boolean'},
                    'args': {'type': 'array'},
                    'creation': {'type': 'string'},
                    'device_class': {'type': ['null', 'string']},
                    'documentation': {'type': ['null', 'string']},
                    'kwargs': {'type': 'object'},
                    'last_edit': {'type': 'string'},
                    'name': {'type': 'string'},
                    'prefix': {'type': 'string'},
                    'type': {'type': 'string'}},
     'required': ['_id', 'prefix', 'type'],
     'type': 'object'}

On instance:
    {'_id': 'alias',
     'active': True,
     'args': [],
     'beamline': 'LCLS',
     'creation': 'Fri Aug 28 10:54:19 2020',
     'device_class': 'types.SimpleNamespace',
     'documentation': None,
     'functional_group': 'FUNC',
     'kwargs': {'hi': 'oh hello'},
     'last_edit': 'Fri Aug 28 10:54:19 2020',
     'location_group': 'LOC',
     'name': 'alias',
     'prefix': 'BASE:PV',
     'type': 'OphydItem',
     'z': 400}
```

@hhslepicka and @klauer  if you have any advice you can give me about using `jsonschema` please feel free to do so, as you both probably worked with this package more than me - I just kind of want to get an idea if this could be a stopper for me somewhere, because as of now I kind of find it a great solution for the `audit` command, but i could be missing something out of lack of knowledge though

Eventually I would have to have a few schemas I check against, depending on the type of the Entry.... and for the `pcdsdevices` I can be specific for each container. ?? maybe

ZLLentz, 2020-08-29:
If we're going this route, I wonder if we want to re-imagine the EntryInfo definitions as jsonschemas? It's certainly a more standard way of doing this... That feels better than duplicating all the EntryInfo definitions, but it also feels like a lot of work

klauer, 2020-08-31:
With the exception of the extra keys, is there a problem going key-by-key and calling `EntryInfo.enforce()` method on the value? I'm not sure jsonschema buys us anything here, except for a long rewrite to get back to what we currently have. Beside for that, the schemas aren't exactly convenient to write, while I think `EntryInfo` is.

In my opinion, supporting extra keys will be something people will continue to want. Whether that comes in the form of top-level object keys as is done currently or, say, nested beneath `"metadata": {...}` is more of an implementation detail than anything else. A detail that could be handled differently on a given backend, if it were to be more strict than mongo/json.

cristinasewell, 2020-08-31:
I see what you mean, thank you for your insight on that - i'll stay away from the `jsonschema` in this

hhslepicka, 2020-08-31:
Extra keys helped us in the past before adding data into a specific container or expanding a container. I agree with @klauer on that and I would also move away from `jsonschema` to avoid duplication of effort with the definition of the EntryInfo and jsonschema.

cristinasewell, 2020-09-02:
For `device_class` - is it guaranteed that it will be of format `somethi.somethingElse.SomeClass` or at least have a `.` in it ???
Also, based on the `loader` - the `device_class` seems to be mandatory, is that the case? If this is not the case, what should the loader do when we encounter a `device_class` that is None (default)??

ZLLentz, 2020-09-02:
I'm struggling to imagine a case where `device_class` doesn't have an `.` in it. At some point there was a mechanism in `happi` where you could configure `somthi.somethingElse` as a standard path for classes, but I'm not sure where that was or if it still exists.

`device_class` definitely should be mandatory, and a `None` value should be an error

cristinasewell, 2020-09-02:
I'm going to keep posting questions here - sorry about that but I run into little things that I don't fully understand. 

1.  we currently don't enforce the `device_class` to be mandatory at `EntryInfo` - should I make it Not Optional? and should I also enforce the format to contain a `.` in it (or is it not necessary) - the reason I ask about this is that in the loader we automatically assume it has a `.` and we don't check if it does not, but we can handle this it in the loader too. 

2. Here is one more thing, is `OphydItem` supposed to have the same type of `args` and `kwargs` as the `HappiItem` ?? 
If i have the default ones for `OphydItem` `['{{prefix}}']` and `{'name': '{{name}}'}` and try to load this, it will complain that the constructor only takes 1 argument: `__init__() takes 1 positional argument but 2 were given` 
This happens I think because in the `loader` we try to instantiate the object with both args and kwargs: ` obj = cls(*args, **kwargs)` but the constructor takes only `**kwargs` (`def __init__(self, **kwargs):`)
What is the approach here, do we want to change the constructor/add one more constructor or to handle this in the loader? Or should i not worry about this one in this ticket??? I could simply return an error for now...
I think the same thing goes for the `HappiItem` but the defaults for this are **empty** list and dictionary so if we use the defaults it won't complain for this one

ZLLentz, 2020-09-03:
1. I'm surprised that we don't require `device_class`... I'm not sure I can imagine a drawback to making it required. I think it is good to require it to have a `.` if we make that assumption later, but I don't think it'll be sufficient to stop bad class names from making it into the database.
2. I don't fully understand this, I expect basically all ophyd devices to have a positional arg called `prefix`. It is ok to require device classes with nonstandard arguments to reflect this in their database entry. What class is failing to load with a `prefix` arg?

cristinasewell, 2020-09-03:
1. Ok thank you, I will make the `device_class` required. As of `.` - i agree with you, this will not necessarily stop us from inserting a bad class names, i'll leave that as `str` for now.

2. Maybe this is because I don't know what I am doing hahah: basically I try to create an Ophyd Item with `happi.Device` for `device_class` - this is probably where i go wrong ... Note that I use both defaults for `args` and `kwargs` - Please let me know if I am doing this wrong :) 
```
(python3.7) cristina@...:~/workspace/happi$ happi add
[2020-09-02 17:42:36] - INFO -  Please select a container, or press enter for generic Ophyd Device container:
OphydItem
HappiItem
pcdsdevices.happi.containers.Acromag
pcdsdevices.happi.containers.AreaDetector
pcdsdevices.happi.containers.Attenuator
pcdsdevices.happi.containers.BeamControl
pcdsdevices.happi.containers.Diagnostic
pcdsdevices.happi.containers.GateValve
pcdsdevices.happi.containers.IPM
pcdsdevices.happi.containers.LCLSItem
pcdsdevices.happi.containers.LODCM
pcdsdevices.happi.containers.Motor
pcdsdevices.happi.containers.MovableStand
pcdsdevices.happi.containers.OffsetMirror
pcdsdevices.happi.containers.PIM
pcdsdevices.happi.containers.PulsePicker
pcdsdevices.happi.containers.Slits
pcdsdevices.happi.containers.Stopper
pcdsdevices.happi.containers.Trigger
pcdsdevices.happi.containers.Vacuum
OphydItem
[2020-09-02 17:42:40] - INFO -  Enter value for name, default=None, enforce=re.compile('[a-z][a-z\\_0-9]{2,78}$')
ophyd_name
[2020-09-02 17:42:46] - INFO -  Enter value for device_class, default=None, enforce=<class 'str'>
happi.Device
[2020-09-02 17:42:55] - INFO -  Enter value for args, default=['{{prefix}}'], enforce=<class 'list'>

[2020-09-02 17:42:58] - INFO -  Selecting default value ['{{prefix}}']
[2020-09-02 17:42:58] - INFO -  Enter value for kwargs, default={'name': '{{name}}'}, enforce=<class 'dict'>

[2020-09-02 17:43:00] - INFO -  Selecting default value {'name': '{{name}}'}
[2020-09-02 17:43:00] - INFO -  Enter value for active, default=True, enforce=<class 'bool'>

[2020-09-02 17:43:07] - INFO -  Selecting default value True
[2020-09-02 17:43:07] - INFO -  Enter value for documentation, default=None, enforce=<class 'str'>

[2020-09-02 17:43:09] - INFO -  Selecting default value None
[2020-09-02 17:43:09] - INFO -  Enter value for prefix, default=None, enforce=<class 'str'>

[2020-09-02 17:43:10] - INFO -  Not an optional field!
[2020-09-02 17:43:10] - INFO -  Enter value for prefix, default=None, enforce=<class 'str'>
PREFIX
[2020-09-02 17:43:13] - INFO -  Please confirm the following info is correct:
+---------------+----------------------+
| EntryInfo     | Value                |
+---------------+----------------------+
| active        | True                 |
| args          | ['{{prefix}}']       |
| device_class  | happi.Device         |
| documentation | None                 |
| kwargs        | {'name': '{{name}}'} |
| name          | ophyd_name           |
| prefix        | PREFIX               |
+---------------+----------------------+
y/N
y
[2020-09-02 17:43:14] - INFO -  Adding device
[2020-09-02 17:43:14] - INFO -  Storing device OphydItem (name=ophyd_name) ...
[2020-09-02 17:43:14] - INFO -  Adding / Modifying information for ophyd_name ...
[2020-09-02 17:43:14] - INFO -  HappiItem OphydItem (name=ophyd_name) has been succesfully added to the database
(python3.7) cristina@....:~/workspace/happi$ happi load ophyd_name
[2020-09-02 17:43:32] - INFO -  Creating shell with devices ['ophyd_name']
[2020-09-02 17:43:32] - ERROR -  Could not load device ophyd_name, __init__() takes 1 positional argument but 2 were given
```

The ERROR above is something I am catching from the `cli.py` when loading.

ZLLentz, 2020-09-03:
Ah! `happi.Device` is actually deprecated for this reason, the name is extremely confusing. See #124, which was never fully completed.

Instead of calling these happi devices, we now call them happi items to avoid confusion with the ophyd Device class. So here you're actually creating a happi item that creates a happi item, and happi items don't have a `prefix` arg.

You'll want to be using `ophyd.Device` (and any subclass of that) as an example valid `device_class` argument for `OphydItem`

See https://github.com/bluesky/ophyd/blob/bd9acd62eeeb837ad11ba399e806c35317559a6e/ophyd/device.py#L653

cristinasewell, 2020-09-03:
Thanks a lot, that makes sense and it works :)

tangkong, 2022-10-14:
Closed by #286

closed 2022-10-14 by tangkong