pcdshub/happi #146: No beamline field during happi add

ZryletTC, 2020-06-24:
## Expected Behavior
`happi add` prompts for which beamline the device is on

## Current Behavior
It doesn't. Resulting database entry doesn't have a `beamline` field:
```
+---------------+--------------------------------------+
| EntryInfo     | Value                                |
+---------------+--------------------------------------+
| active        | True                                 |
| args          | ['{{prefix}}']                       |
| creation      | Wed Jun 24 15:32:26 2020             |
| device_class  | pcdsdevices.sample_delivery.Selector |
| documentation | None                                 |
| kwargs        | {'name': '{{name}}'}                 |
| last_edit     | Wed Jun 24 15:35:02 2020             |
| name          | icl_sds_sel_b                        |
| prefix        | ICL:SDS:SEL:B                        |
| type          | OphydItem                            |
+---------------+--------------------------------------+
```

Somewhat related: When I run `happi edit icl_sds_sel_b beamline=ICL`, the record doesn't seem to change. I would expect it to add the beamline field with the requested value.

## Context
I'm trying to add ICL sample delivery devices to the happi database.

## Your Environment
pcds-3.3.0 on psbuild-rhel7

ZLLentz, 2020-06-24:
`OphydItem` does not have `beamline` because not all `ophyd` devices are required to be part of a `beamline`.

Try using `LCLSItem` instead. This will have additional LCLS-specific fields added.

hhslepicka, 2020-06-24:
@ZryletTC I think it is due to the type being `OphydItem` and not `pcdsdevices.happi.containers.LCLSItem`.
OphydItem is the most basic form that does not require the LCLS metadata.

ZryletTC, 2020-06-24:
Ahh 👍 thank you!
Are the fields specified by the container and can't be added to?

hhslepicka, 2020-06-24:
dang it... I forgot to press send! :(

ZLLentz, 2020-06-24:
Fields are specified by the container so that you can't accidentally add misspelled variants of the fields. I'd consider `happi edit` failing silently in the case above to be a bug.

ZryletTC, 2020-06-25:
Make sense, I'll make separate issue for that bug.

closed 2020-06-25 by ZryletTC