pcdshub/happi #162: Lack of some coverage/tests for CLI

cristinasewell, 2020-08-19:
## Expected Behavior

Some testing for the `add`, `clone`, `edit`, `load` commands would be great.

## Current Behavior

Currently those commands are not tested 
## Possible Solution

https://github.com/pcdshub/happi/pull/161#issuecomment-676606588
## Steps to Reproduce (for bugs)

## Context

 This would allow programmers to refactor code and make sure the module still works correctly.

## Your Environment

ZLLentz, 2020-08-19:
This is probably my mistake, we were in a rush to get these out and never revisited them for testing.

cristinasewell, 2020-08-19:
@ZLLentz  can I try to work on this Issue?

ZLLentz, 2020-08-19:
I would be very :+1: on filling the testing gaps, especially here in CLI, a very common user codepath

PR #164 (2020-08-26): TST: Cli testing

## Description

Added more test cases for some cli functionalities 
## Motivation and Context

Trying to close #162 

## How Has This Been Tested?

```
Has only been tested locally: 

python run_tests.py
coverage run run_tests.py 
coverage report -m
```
## Where Has This Been Documented?

N/A
closed 2020-08-26 by klauer