pcdshub/happi #165: Identify Issues with CLI Tools

ZLLentz, 2020-08-26:
## Expected Behavior

Users should use the CLI or the IPython interface for adding devices to and editing devices in `happi` to take advantage of schema checking and to avoid corrupting the database files.

## Current Behavior

Many users ignore this and edit the json files directly.

## Possible Solution

Figure out why the existing tools are hard to use, document the findings, and propose changes.

ZryletTC, 2020-09-09:
For one thing, why aren't all commands listed in the usage output?

ZryletTC, 2020-09-09:
I also resort to json editing sometimes because the CLI `add` function doesn't allow me to set some values like I want them. For example, I think it won't let me set 'args' or 'prefix' to be blank.

klauer, 2020-09-19:
Issue @ZryletTC just found:

```
[2020-09-18 17:55:14] - INFO -  Enter value for kwargs, default={'name': '{{name}}'}, enforce=<class 'dict'>
{'name': '{{name}}', 'n_bounces': 10}
[2020-09-18 17:56:03] - INFO -  Invalid value {'name': '{{name}}', 'n_bounces': 10}
```

`dict`-enforced items fail on the CLI as `dict('{}')` doesn't work.  We should probably be using something like:

```
In [5]: ast.literal_eval("{'name': '{{name}}', 'n_bounces': 10}")
Out[5]: {'name': '{{name}}', 'n_bounces': 10}

In [6]: dict(_)
Out[6]: {'name': '{{name}}', 'n_bounces': 10}
```