pcdshub/happi #167: JSON load failure "expecting property name enclosed in double quotes"

klauer, 2020-09-03:
[image]

Guesses:
* There may have been a write in process
* NFS may have given out while reading it

Should happi json backend retry loading in the event of a syntax error?

ZLLentz, 2020-09-03:
Should happi json backend open the file and show us line 3977 in cases like this?

klauer, 2020-09-03:
It looks like just bad timing, unless someone was really editing the JSON directly:
[image]

Showing where the bad line is would help developers, but making another attempt to load the file could help users. 🤷‍♂️

ZLLentz, 2020-09-03:
Try three times, then show the line?

I agree the saving could be safer. We also may be getting closer to wanting split files for different areas.

ZryletTC, 2020-09-03:
Sorry, that was me

> It looks like just bad timing, unless someone was really editing the JSON directly

Both, bad timing as I was writing to it directly and introduced a bug for like five minutes.

> Should happi json backend open the file and show us line 3977 in cases like this?

Yes, please :)

> but making another attempt to load the file could help users

Also yes please :)

klauer, 2020-09-09:
Since we identified that it wasn't happi writing the file or failing to read it, this issue isn't exactly valid.

However, other good discussions regarding the root cause of this and maybe potential paths forward are popping up (#165, #169, among others).

closed 2020-09-09 by klauer