pcdshub/happi #183: Custom args/kwargs needs better handling

ZLLentz, 2020-11-04:
## Expected Behavior

It should be easy to add a device that passes in extra args/kwargs.

## Current Behavior

Very awkward when using happi add, not clear if it even interprets the input correctly.

## Steps to Reproduce (for bugs)

1. `happi add`
2. change args or kwargs
3. your device is probably broken now

klauer, 2020-11-04:
Related: https://github.com/pcdshub/happi/issues/165#issuecomment-695141777

ZryletTC, 2021-12-09:
Just encountered this again
```
(pcds-5.1.1) psbuild-rhel7-01:R1.0.16 pennebak$ happi edit xcs_sb1_ipm "kwargs={'name': '{{name}}', 'prefix_wave8':'XCS:SB1:BMMON'}"
[2021-12-08 23:24:38] - INFO -  Setting xcs_sb1_ipm.kwargs = {'name': '{{name}}', 'prefix_wave8':'XCS:SB1:BMMON'}
[2021-12-08 23:24:38] - ERROR -  Could not edit xcs_sb1_ipm.kwargs: dictionary update sequence element #0 has length 1; 2 is required
```
Also confused by why the following results in two setattrs. Looking at https://github.com/pcdshub/happi/blob/master/happi/cli.py#L241, I would expect that even if the two elements got split into multiple 'edits', the value of field wouldn't be 'kwargs' for both.
```
(pcds-5.1.1) psbuild-rhel7-01:R1.0.16 pennebak$ happi edit xcs_sb1_ipm kwargs={'name':'{{name}}','prefix_wave8':'XCS:SB1:BMMON'}
[2021-12-08 23:25:05] - INFO -  Setting xcs_sb1_ipm.kwargs = name:{{name}}
[2021-12-08 23:25:05] - ERROR -  Could not edit xcs_sb1_ipm.kwargs: dictionary update sequence element #0 has length 1; 2 is required
[2021-12-08 23:25:05] - INFO -  Setting xcs_sb1_ipm.kwargs = prefix_wave8:XCS:SB1:BMMON
[2021-12-08 23:25:05] - ERROR -  Could not edit xcs_sb1_ipm.kwargs: dictionary update sequence element #0 has length 1; 2 is required
```

ZLLentz, 2021-12-09:
I think there must be something screwy with how the unquoted curly brackets are being interpreted in bash in your second example

klauer, 2021-12-09:
I think there are at least two parts to it. 
1. The shell interpretation of the text (which might be resolved by using something like `happi edit xcs_sb1_ipm kwargs="{'name':'{{name}}','prefix_wave8':'XCS:SB1:BMMON'}"` - with outer double quotes)
2. But primarily the need for a `literal_eval` to make it into an _actual_ dictionary. While `enforce=dict` on the `kwargs` part of the container, but calling e.g. `dict("{}")` would raise an exception.