pcdshub/happi #187: happi edit command not working for boolean variables

Mbosum, 2020-11-18:
## Expected Behavior
happi entry for <motor_name>:
.
some_boolean=False
.
(pcds-3.4.1) xcs-control $ happi edit <motor_name> some_boolean=True
some_boolean=True

**or reverse:** 
happi entry for <motor_name>:
.
some_boolean=True
.
(pcds-3.4.1) xcs-control $ happi edit <motor_name> some_boolean=False
some_boolean=False

or:
happi entry for <motor_name>:
.
some_boolean=False

(pcds-3.4.1) xcs-control $ happi edit <motor_name> some_boolean=False
some_boolean=False

## Current Behavior
happi entry for <motor_name>:
.
some_boolean=False

(pcds-3.4.1) xcs-control $ happi edit <motor_name> some_boolean=False
some_boolean=True

**or reverse:** 

happi entry for <motor_name>:
.
some_boolean=True
(pcds-3.4.1) xcs-control $ happi edit <motor_name> some_boolean=False
some_boolean=True

**An example in the field of this is** 
happi entry for xpp_pol_wp:
.
lightpath=False
.
(pcds-3.4.1) xcs-control $happi edit xpp_pol_wp lightpath=False
lightpath=True

**or:**
happi entry for xpp_pol_wp:
.
active=True
(pcds-3.4.1) xcs-control $happi edit xpp_pol_wp active=False
active=True

ZLLentz, 2020-11-19:
Did a quick readthrough of the code, at first look it seems like `happi` is simply casting the string "False" to a boolean... which is `True`:
```
In [2]: bool("False")
Out[2]: True
```
Because boolean fields are implemented like so:
https://github.com/pcdshub/pcdsdevices/blob/c5d9806d11a680ea1985f651ce2ecfa0df3b299f/pcdsdevices/happi/containers.py#L25

And the code simple calls the `enforce` function as a preprocessor on the user input.

This suggests that you can do the following as a workaround:
`happi edit xpp_pol_wp lightpath=`

Which should put an empty string into the database- a `False` value. (I have not tried this yet, but that's the theory).

To resolve this issue we probably need to rework the value enforcement for boolean fields a bit, or carve out a special case for them in the edit cli.

klauer, 2020-11-19:
Users seem to be running into these pretty consistently: https://github.com/pcdshub/happi/issues/165#issuecomment-695141777

Should probably make a point of reviewing all keys to see if this list is exhaustive.

PR #212 (2021-02-10): MNT: fix confusing behavior with boolean enforcement

## Description

Add a special case for boolean entry enforcement where we interpret the input string instead of just casting it into a boolean.
Accepts strings like "true" and "false" as stand-ins for `True` and `False` in situations where the input must be given as a string, such as in the cli.

## Motivation and Context

`happi edit` and `happi add` do not set boolean values as one would expect
closes #187

## How Has This Been Tested?

Added tests

## Where Has This Been Documented?

Will include in release notes, I'm going to count this [... cut]
closed 2021-02-10 by ZLLentz