pcdshub/happi #189: default backend path

untzag, 2020-12-21:
## Expected Behavior

As a brand new happi user, I'm surprised that there is no default location for the database backend. Instead, users are forced to provide a database path every time that they interact with happi. I believe that a sane default location could reduce the friction of interacting with happi.

## Current Behavior

Users must already know the path to the happi database backend.

## Possible Solution

I propose you use the [appdirs](https://pypi.org/project/appdirs/) package. This package abstracts away differences between different operating systems.

```python
>>> import appdirs
>>> import pathlib
>>> pathlib.Path(appdirs.user_data_dir("happi")) / "db.json"
PosixPath('/home/blaise/.local/share/happi/db.json')  # on my machine
```

If the file does not exist, happi should silently initialize it. In this way, users can get started very easily:

```python
import happi
hc = happi.Client()  # uses default backend, location
```

## Steps to Reproduce (for bugs)

N/A

## Context

I'm using happi to help expose [yaq](https://yaq.fyi/) devices to clients via [yaqc-bluesky](https://github.com/bluesky/yaqc-bluesky). These are simple systems where users expect to walk up and quickly interact with all of the relevant devices. A variety of users share the same instrument, so one user may not know where the database is. These users are working with graphical clients but also in various REPLs.

This might be a bad idea for some reason, but as a new happi user I was surprised :smile: 

## Your Environment

N/A

klauer, 2021-01-04:
There is room for improvement here, certainly. Agreeing on the path forward will probably be the most difficult part.

Currently, happi uses the `HAPPI_CFG` environment variable or the POSiX-centric `XDG_CONFIG_HOME` (falling back to `~/.config`) to specify the database configuration (https://github.com/pcdshub/happi/blob/fbbc8a645ad99fc9f46f4fa3d17e0dcf1663156d/happi/client.py#L691-L696).

For our usage, `HAPPI_CFG` gets set in our user-facing conda environments activation scripts - requiring no setup on their part. Would something similar not work in your case?

Implicitly creating a database seems like it could easily go either in the direction of "great feature" or "great source of confusion". If an environment setting or config file isn't set appropriately, and a user searches for their device, the database gets created, and the user sees "No matching device found!" and a bug report gets filed.

Perhaps an alternative, less implicit, would be something along the lines of:

```python
if check_if_interactive_tty() and no_database_found:
    if ask_user_to_create_default_database():
        create_default_database()
    else:
        sys.exit(1)
```

Then, at least, new users could create an empty database (defaulting to JSON backend) without much trouble? 

I also feel this engineer/operator-focused tool should be configured by those who know its appropriate settings for the given facility, and somewhat feel the above is unnecessary. I'm open to opinions, in any case.

untzag, 2021-01-24:
I now think my original suggestion was misguided. Closing.

closed 2021-01-24 by untzag