pcdshub/happi #198: confusing behavior at import time for packages implementing entry points

untzag, 2021-01-08:
## Expected Behavior

The happi registry leads to some very confusing behavior when trying to load HappiItem subclasses using entry points. The surface-level observation is that the registry seems to be import-order dependent. I'm pretty sure I understand exactly why this happens---only question is how to fix it :thinking:.

## Current Behavior

You can actually cause unexpected behavior using your own pcdsdevices package.

This following works as expected:

```python
>>> import happi
>>> from pcdsdevices.happi import containers
>>> happi.containers.registry._registry   # contains pcdsdevices
```

The following does NOT work as expected:

```python
>>> from pcdsdevices.happi import containers
>>> import happi
>>> happi.containers.registry._registry   # does NOT contain pcdsdevices
```

## Possible Solution

Unexpected behavior occurs when happi is _first_ imported by the container-defining module. If this happens, your `HappiRegistry` singleton will instantiate DURING module import, and BEFORE the container classes themselves are defined. Conversely, if happi is imported before Python touches the container-defining module, happi itself will load the module properly within `HappiRegistry.load`, and everything works as expected.

Minimally invasive way to fix this is to have the registry try to load again when an item is not found. I've got a working version of this, PR coming soon.

I will ask---why does the registry work this way? Couldn't we just have items be lazy-loaded on the client level?

## Context

Unlike pcdsdevices, my package yaqc-bluesky intends to define the happi containers as a top-level module that's imported with the top-level package. I guess I don't _have_ to do it this way, but it's my preference. Given this design, I was immediately hit with very confusing import-order dependence between happi and my top-level module. I imagine you may never have to deal with this for pcdsdevices, since the containers are defined within a subpackage.

klauer, 2021-01-08:
Thanks for reporting this. It's definitely a frustrating/confusing bug we should squash. 

Stepping back for a second, the primary reason to load all entries in advance is that a GUI (or TUI) could enumerate the containers that exist when trying to create a new one. This then necessitated the entrypoints addition for ease of use. 

I haven't thought much into the implications just yet, but one possible idea came to mind:

 The entrypoint handles the importing now just fine, so we could then tweak `__init_subclass__` in `HappiItem` (https://github.com/pcdshub/happi/blob/05295b8617e23bb3d2ab27f17dac6ef36008a11b/happi/item.py#L161-L162) to insert the class into the registry immediately at definition. No container classes would be missed, import ordering would not matter, and there wouldn't be a need for `inspect.getmembers` on the module as there is now. The registry would need a mapping of 'module name' to 'entrypoint name' (as there is that - probably undocumented - layer of indirection there to allow for downstream package restructuring without breaking the names of container classes).

In the meantime, though, I think your PR is a reasonable short-term fix. 👍

closed 2021-01-08 by ZLLentz

klauer, 2021-01-08:
I'll pull my comment out into a new issue rather than reopening this.

PR #199 (2021-01-08): FIX: HappiRegistry call load again if requested item not found

## Description

resolves #198 

## Motivation and Context

see #198 

## How Has This Been Tested?

Tested locally. Willing to add tests but I need guidance from maintainers :sweat_smile: 

## Where Has This Been Documented?

Willing to add documentation but I need guidance from maintainers.