pcdshub/happi #201: Refactor container loading mechanism

klauer, 2021-01-08:
## Current Behavior
The entrypoints-based implementation loads entrypoints and scans through the provided modules to find `HappiItems` subclasses.

Depending on import ordering, container classes may not exist at the time the registry is loaded. A short-term fix to get around that was included in #199.

## Expected Behavior
* Container loading should not be dependent on import ordering.
* It should also not be unreasonably slow when trying to load an item that doesn't exist (a potential side effect of #199).
* Scanning through module dictionaries for classes should not be required when `__init_subclass__` exists for this purpose
 
## Possible Solution

* Entrypoint handles the importing and generates a map of module name to entrypoint name
    * Mapping retains original functionality: layer of indirection to allow for downstream package restructuring without breaking the names of container classes
* Tweak `__init_subclass__` in `HappiItem` (https://github.com/pcdshub/happi/blob/05295b8617e23bb3d2ab27f17dac6ef36008a11b/happi/item.py#L161-L162) to insert the class into the registry immediately at definition. 
    * No container classes would be missed
    * Import ordering would not matter
    * There wouldn't be a need for `inspect.getmembers` on the module as there is now.
   
## Context
The primary reason to load all entries in advance is that a GUI (or TUI) could enumerate the containers that exist when trying to create a new one. This then necessitated the entrypoints addition for ease of use (not requiring modules to be imported manually).

ZLLentz, 2021-01-08:
I think `__init_subclass__` usage would be very clean here.

I have some conceptual confusion here regarding entry points, so I'm going to spend some time reading up on them again. Would this force all `HappiItem` subclasses to be included in the registry? Even ones made outside of entry point declarations? (provided that the class statement is executed during some import)

klauer, 2021-01-08:
> Would this force all HappiItem subclasses to be included in the registry? 

Two paths that I see:
1. Add all, even those without entry points. 
    a. If there is no associated entry point, use the fully qualified `module.classname` as the container name. 
    b. In this scenario, there could be an opt-out mechanism, if needed.
2. If there's no entry point designated for a given subclass, skip adding it to the registry. 

I'm leaning toward (1), though (2) may be cleaner.

Happy to resolve any confusion about entry points in Slack - you know where to find me.

ZLLentz, 2021-01-08:
My reading session was fruitful and I understand entry points sufficiently well now.

After some thought, it seems like the only cases where the original implementation fails is when there are circular import shenanigans. And in all of those cases, our classes that need to be added to the registry will be loaded soon afterwards.

(2) would force users to define the entry points, which would alleviate some confusion with cases like "why didn't my special class get included" in cases where their containers file is never imported (and why would it be, if it only contains information for happi).

Another option to consider is (3), add all and show a warning about the missing entry point. "You've created a `HappiItem` subclass without defining an entry point for it, you should define an entry point to make sure it is always loaded."

klauer, 2021-01-08:
Those are good points and are enough to shift my opinion. I like (3) for its adherence to "you must use good practices (make a package and use entrypoints) if you want your containers in the registry", and it tells users what they need to do to fix it.

klauer, 2022-05-06:
## Path forward

1. Add all subclassed containers to the registry, regardless of entrypoint settings
2. Add a warning about containers that are not included in entrypoints (when they are defined or maybe used?)