pcdshub/happi #202: Some dependencies in conda recipe are optional

klauer, 2021-01-19:
## Current Behavior
Some dependencies are optional (such as `psdm_qs_cli`, `pymongo`), but we require them in our conda recipe.

## Expected Behavior
Optional dependencies should not turn into requirements in our conda recipe.

## Possible Solution
Utilize `run_constrained` to constrain versions of these optional dependencies.
https://docs.conda.io/projects/conda-build/en/latest/resources/define-metadata.html#run-constrained

## Context
Recipe: https://github.com/pcdshub/happi/blob/master/conda-recipe/meta.yaml

ZLLentz, 2021-01-19:
I've never seen this, very interesting. Would we want to set these up as such on the conda-forge recipe too?

klauer, 2021-01-19:
Yes, especially on the conda-forge recipe - as it's where people would install happi from primarily. 

I'm thinking when we make this change we can also better leverage conda recipe's support for test dependencies and test scripts - further reworking our CI structure. (as we briefly chatted about on Slack)

closed 2022-05-11 by ZLLentz