pcdshub/happi #204: happi cfg documentation?

untzag, 2021-01-24:
Why do I need a configuration file to use the happi CLI? I don't need one for the Python interface, as far as I can tell. 

Are the contents of this file documented anywhere?

klauer, 2021-01-25:
Most likely your config file is either set by way of `HAPPI_CFG` or you may have put it in `~/.config` as we had talked about in https://github.com/pcdshub/happi/issues/189#issuecomment-754130415.

The happi CLI will try to load from the default locations if `HAPPI_CFG` and `--path` are not specified. The Python `Client` interface will do the same when you call `happi.client.Client.from_config()`.

As to documentation - I think the answer is not yet.  The contents are backend-dependent, as instantiating a JSON "database" client requires a path to a file, whereas the mongo backend requires server information.

Digging into the source
--------
https://github.com/pcdshub/happi/blob/5c7b3a2a94e8beec8927287465bbffcf239b35e5/happi/client.py#L664-L691

Indicates: `DEFAULT` is the required "ini"-like section, and this is where the keywords to the backend initialization are derived from.

JSON backend:  https://github.com/pcdshub/happi/blob/5c7b3a2a94e8beec8927287465bbffcf239b35e5/happi/backends/json_db.py#L49

So here, `path` is the required setting.

Mongo: https://github.com/pcdshub/happi/blob/5c7b3a2a94e8beec8927287465bbffcf239b35e5/happi/backends/mongo_db.py#L40-L42

For this one, despite being optional in the initializer, the format string above pretty clearly shows you'll need to have `user`, `pw`, `host`, and `db`.

untzag, 2021-01-26:
Thanks for your patient response @klauer. I now see exactly how the config file works. I'll write some notes to myself about it :smile:

klauer, 2021-01-26:
No problem at all. I'll flip that around and say thanks for prodding us to make happi better.

This is still a valid issue - so I'll reopen it as a reminder that we need to actually document this.

tangkong, 2022-10-06:
With the addition of #287 , there's some notes to be added about the multi-backend backend, particularly when it comes to backend priority and config file section ordering.