pcdshub/happi #206: HappiItem rejects names under two charachters

untzag, 2021-01-24:
## Expected Behavior

## Current Behavior

## Possible Solution

## Steps to Reproduce (for bugs)

1.
2.
3.

## Context

## Your Environment

klauer, 2021-01-25:
For our purposes, the happi database holds a huge number of items and is shared among many different hutches/beamlines. Choosing a one or two character name will only serve to cause confusion or potential for clashing names. For our users who insist that typing more than 2 characters is just _too many_, we suggest they configure aliases in their beamline-specific startup scripts.

This may be something we have to revisit, as not all facilities (or users such as yourself) will have the same constraints. Naming requirements should be facility-specified and not forced on you by happi.

That said, I think we can all agree that at minimum `name.isidentifier()` should be `True`.

untzag, 2021-01-25:
Y'all already raise an exception if a user accidentally tries to overwrite an existing `_id`.

klauer, 2021-01-25:
> Y'all already raise an exception if a user accidentally tries to overwrite an existing _id.

Indeed, we do. "First come, first serve" sadly is insufficient for facility naming standards.

untzag, 2021-01-25:
That makes sense. Perhaps y'all could be permissive by default but respect an environment variable to lock down name options when appropriate.

ZLLentz, 2021-01-25:
I think it would be ok to open up the tight restrictions of the `name` entry on `HappiItem`. LCLS can easily restrict the database names in our site-specific containers, as can any other site that wants to go that way for the reasons @klauer explained above. Another option is to include a tightly-restricted `OphydItem` subclass here with some "common restrictions" on names, etc., but I'm not sure how useful that is.

PR #211 (2021-02-10): ENH: allow short and long names

## Description

- Remove the requirement for names to be between 3 and 80 characters
- Keep requirement for names to be valid Python identifiers
- Add previously implicit requirement for names to not conflict with reserved keywords
- Add feature to use arbitrary functions as enforce parameters

## Motivation and Context

This is a site-specific assumption that should not be baked in to the base class
closes #206 

## How Has This Been Tested?

~Online regex tester site to quickly vet python identifier~
The above is no longer valid, since we are no longer using  [... cut]
closed 2021-02-10 by ZLLentz