pcdshub/happi #220: Remove fcntl warning

klauer, 2021-05-17:
The following message on Windows is annoying:

https://github.com/pcdshub/happi/blob/d5cba946cd4a58a0b6a9da011e256f84d9ced81a/happi/backends/json_db.py#L22

This is something that should be part of the documentation, not a message on every invocation of `happi`.

klauer, 2021-05-19:
Remove backend unavailable warnings, too?

JulReinhardt, 2021-05-19:
I agree that the fcntl warning is not necessary. You want to remove backend warnings, like when json or mongo packages are not properly installed?

klauer, 2021-05-19:
Yes, @JulReinhardt, I think it should only be an error if you actually try to _use_ those backends when they are unavailable. Seem reasonable?

JulReinhardt, 2021-05-19:
Yes, agreed.

PR #237 (2022-03-11): MNT: switch fcntl warning to debug

## Description
* Reduce on-import warning of fcntl to debug.
* This is shown repeatedly on Windows for ads-deploy - which doesn't even really use happi - and is a bit annoying.

## Motivation and Context
Closes #220 

## How Has This Been Tested?
If it passes flake8 and imports, this should be fine.

## Where Has This Been Documented?
Issue / PR text
closed 2022-03-11 by klauer