pcdshub/happi #221: Happi Front-end

ZLLentz, 2021-08-05:
## Expected Behavior

Develop a GUI that lets you browse and edit the happi database

klauer, 2021-08-05:
Web-based or PyQt-based are our top options.

ZLLentz, 2021-08-05:
Web-based:
- happi server process (with some web auth?)
- web front-end that communicates with the process

PyQt-based:
- local application that runs qt and happi
- auth is based on being able to log in to the system

klauer, 2022-05-06:
Some browsing support is available already:
https://github.com/pcdshub/happi/blob/master/happi/qt/model.py

atef improves upon this with a happi item + device/component browser. Some of this should be upstreamed to happi:

[image]

tangkong, 2023-11-07:
Should we close this and make more specific issues?  top of mind thanks to #246 
- a web gui might not be absurd once we properly switch to mongo
- editable via gui might be in striking distance after merging in atef changes

klauer, 2023-11-07:
Sounds good to me 👍

closed 2023-11-07 by klauer