pcdshub/happi #224: CI Docs failing to build

ZLLentz, 2021-08-19:
## Current Behavior

```
Warning, treated as error:
/home/travis/build/pcdshub/happi/docs/source/client.rst:33:more than one target found for 'any' cross-reference 'Client': could be :py:class:`happi.Client` or :py:class:`happi.client.Client`
Makefile:24: recipe for target 'html' failed
make: *** [html] Error 2
```

ZryletTC, 2021-08-20:
This is an easy fix but seems to have been triggered by using the new version of sphinx. Sphinx 4.1.2 shows this warning/error but version 3.5.4 succeeds without any warnings. I don't see anything in their changelog that would explain this but they must've made it stricter about ambiguous cross-reference targets somewhere.

ZryletTC, 2021-08-20:
If you agree, I'm gonna stop digging into where in their code it became stricter and just make the fix in our code.

ZLLentz, 2021-08-20:
Fixing our docs or forcing sphinx to the old version are both acceptable solutions

PR #226 (2021-08-21): BUG: Fix ambiguous cross-reference to Client

## Description
What the title says. Also changed the allow failure name to match pcdshub/pcds-ci-helpers#65

## Motivation and Context

Fixes #224  

## How Has This Been Tested?

Built docs locally, hoping Travis approves as well.

## Where Has This Been Documented?

Nowhere.
closed 2021-08-21 by ZLLentz