pcdshub/happi #231: heuristic loading

untzag, 2021-12-22:
## Expected Behavior

Would be nice for me if `happi load "*"` worked, in analogy to `happi search "*"`.

## Current Behavior

`happi load` requires users to explicitly specify which hardware to load. None of the search functionality is supported.

## Possible Solution

Would be nice to provide support for searching and loading in one user-friendly expression.

untzag, 2022-01-03:
The obvious idea is just to route the arguments to `load` through the same code path as `search`.

Problem is that `happi search <criteria>` combines multiple criteria with AND. So current syntax of `happi load name1 name2 name3` would result in nothing loaded because no items have all three of those names. `load` needs OR logic to work as a user expects, but then it becomes impossible to narrow down to a specific item based on multiple criteria. Maybe the OR logic would only apply to names?

Perhaps #232 is the best answer after all :thinking:. My only problem is that most users don't really understand the shell enough to use tricks like that (I myself mess up all the time...). Simply dropping into a repl with your objects is something I want everyone to be comfortable with.

@klauer @ZLLentz tell me what to do please. I'll implement it and make a PR, I'm just not sure what the right approach is.

ZLLentz, 2022-01-03:
You're right, the OR vs AND conflict here makes this tricky, and most solutions would probably make the CLI confusing (which is a problem, since the point of this is to make the invocation more user-friendly).

I'd like to hear Ken's opinion on this when he's back from his vacation. My initial thought would be to add some syntax like:
```
happi load --search "*"
```
And we'd be able to compose multiple of them like:
```
happi load name1 name2 --search name="*_mot" --search class="*Motor"
```
Where the above would load in the devices named `name1` and `name2`, all devices whose names end in `_mot`, and all devices whose classes end in `Motor` (and only one of each device if they show up in multiple searches)

But looking at the options in argparse this seems awkward for more complicated searches...

klauer, 2022-01-06:
To get a truly flexible query, we'd need some sort of `x AND (y OR NOT z)`-style grammar which would be rather excessive (in my opinion) for a tool like happi. I have yet to see a nice implementation of this sort of thing in a CLI tool - see `find` for a quick example, with its chainable expressions by way of `-and` and `-or`.

@ZLLentz 's suggestion above seems fine, though the implicit "OR" between the names/`--search` items may not be desirable. 

I think the real question here is what will the most common use cases here be? And how can we cover that with minimal added complexity?

Adding some random thoughts onto Zach's above; we could allow a switch to `OR` mode by way of `--or`:
```bash
$ happi load "name*" --or --search name="*_mot" class="*Motor"
$ happi load --and --search name="*_mot" class="*Motor"
```

Alternatively perhaps `--and`/`--or` could take expressions as in `find`:
```bash
$ happi load "name*" --and class="*Motor"
$ happi load "name*" --or class="*Motor"
```

Then if you need anything beyond that, please use Python? (My brain is still warming up after the winter break, so take the above with a grain of salt...)

untzag, 2022-01-06:
Actually your suggestions convince me that composing search(es) and load together is actually the best approach, as in #232. For example if I want a particular sensor and all of my motors: 

``` bash
$ happi load $(happi search "mysensor" --names) $(happi search class="*Motor" --names)
```

Or if I want to do something repetitive I can even write the names I care about to a file and `cat` into `load`. I think that can cover any "advanced" case I could imagine.

To be honest the only syntax I really have a problem with is how awkward it is to simply load everything. Perhaps just special casing `"*"` is all I actually want. Of course my needs are very simple compared to actual beamlines :-)

klauer, 2022-01-07:
That seems fine to me - I'm not opposed at all to that level of shell command composition, myself. Even for the average user a couple examples in the documentation should be sufficient.

I'm fully in support of reusing the search syntax for `load`, by the way, since it appears I didn't mention that above.

klauer, 2022-05-06:
Related: https://github.com/pcdshub/happi/issues/172

Marking this as 'help wanted' for next week's codeathon.

closed 2025-01-07 by tangkong