pcdshub/happi #279: Importing containers can result in conflicts during entry point containercollection

tangkong, 2022-07-12:
The conflict shows itself as:
``` 
(pcds-5.4.1)roberttk@psbuild-rhel7-01:~/devrepos/pcdsdevices(enh_lp_mixin +)$ happi container-registry
File "/cds/home/r/roberttk/devrepos/happi/happi/containers.py", line 208, in load
    self._safe_add(entry_name, var)
  File "/cds/home/r/roberttk/devrepos/happi/happi/containers.py", line 173, in _safe_add
    raise RuntimeError(f"Duplicated entry found. Keys: {key} "
RuntimeError: Duplicated entry found. Keys: pcdsdevices.happi.containers.LightpathItem and lightpath.happi.containers.LightpathItem point to same class: <class 'lightpath.happi.containers.LightpathItem'>
```

When you import `from lightpath.happi.containers import LightpathItem` in `pcdsdevices.happi.containers`

It seems the container gets collected once in lightpath and again in pcdsdevices under different keys, since they're being collected from different entrypoints.  I can add an issue for it

_Originally posted by @tangkong in https://github.com/pcdshub/lightpath/pull/133#discussion_r919310984_

klauer, 2022-07-12:
I gave this a bit more thought there are a couple ways you could resolve the initial problem without changes to happi:
1. Adjust the import style in lightpath
  a. `happi` uses `inspect.getmembers()` on the module 
  b. So, if the conflicting container type is not found in - effectively `globals()` - then it should not be found by that mechanism
  c. Then: `from lightpath.happi import lightpath_containers` and `LCLSLightPathItem(lightpath_containers.LightpathItem)`
2. Much less preferably: we could be more specific in pcdsdevices and only export happi container classes by name (you have the choice of [class](https://github.com/pcdshub/happi/blob/f348c1ae9dda82245a631af802d56a766b2e0228/happi/containers.py#L204) or [module](https://github.com/pcdshub/happi/blob/f348c1ae9dda82245a631af802d56a766b2e0228/happi/containers.py#L206) in the entrypoint)

A broader question would be: "is it OK to have aliases to containers?"
I think I'd like to continue to say that the answer is "no", because it could cause confusion down the line, make finding items of the same type more difficult, and so on. This is definitely open for discussion.

klauer, 2022-07-14:
Thoughts, @tangkong?

tangkong, 2022-07-14:
I totally thought I left a comment here whoops

I think 1 is acceptable, and is probably the simplest solution.  It will be an additional thing to communicate/document, since it's not obvious that you can't just import the class you need to inherit from it.  (Every package in the future that wants to expose inheritable containers will have to do this)

Could we change how we handle collisions?  The first thought is to just ignore the duplicate entry, but I'm not sure if the order packages are collected in is consistent, which might make the key indeterminate.  We could set up some logic (take the name that's first in alphabetical order?) to make it consistent I guess.  

I agree that aliases to containers (several keys to the same container) will lead to confusion

I think on balance we can probably go with 1 in the short term, and is probably what we should do until we find a more elegant solution (if there is one)

tangkong, 2022-07-14:
Agreed on all points.  I'm coming to terms with (1) being simple enough as long as people read the documentation we write. 😬