pcdshub/happi #299: Invalid containers can break `dict(client)` call

klauer, 2023-02-17:
## Current Behavior
Calling `dict(client)` when an invalid container is in the database raises an exception (and, of course, stops it from returning even a partial dictionary).

```python
$ python -m whatrecord.plugins.happi
Traceback (most recent call last):
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/site-packages/happi/client.py", line 305, in _get_item_from_document
    item = self.create_item(doc["type"], **doc)
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/site-packages/happi/client.py", line 239, in create_item
    raise TypeError(
TypeError: The container class 'pcdsdevices.happi.containers.Leviton' is not in the registry

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/site-packages/happi/client.py", line 491, in __getitem__
    item = self._get_item_from_document(self.backend.get_by_id(key))
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/site-packages/happi/client.py", line 307, in _get_item_from_document
    raise EntryError(
happi.errors.EntryError: The information relating to the container class has been modified to the point where the object can not be initialized, please load the corresponding document

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/site-packages/whatrecord/plugins/happi.py", line 400, in <module>
    _cli_main()
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/site-packages/whatrecord/plugins/happi.py", line 393, in _cli_main
    results = main(**vars(args))
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/site-packages/whatrecord/plugins/util.py", line 35, in wrapper
    results = func(*args, **kwargs)
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/site-packages/whatrecord/plugins/happi.py", line 352, in main
    for item in dict(client).values()
  File "/cds/group/pcds/pyps/conda/py39/envs/pcds-5.6.0/lib/python3.9/site-packages/happi/client.py", line 493, in __getitem__
    raise KeyError(key) from ex
KeyError: 'cxi_leviton_r51_h'
```

## Possible Solution
Be more lenient about bad containers, perhaps provide information on failures even

## Context
`happi-to-confluence`

## Your Environment

pcds-5.6.0
whatrecord 0.4.2
happi 2.0.0

tangkong, 2025-05-30:
Looking at this, do we actually want to be more permissive here?  It feels weird to make dict access (`client["my_dev"]`) simply return none for malformed entries instead of raising.  

I looked at the `dict` function and it simply iterates through the object and tries to __getitem__ each key.  So it's not quite possible to have the individual access raise (`client["malformed"] -> KeyError`) while having dict construction simply skip malformed entries (`dict(client)`)

@ZLLentz any thoughts here?

ZLLentz, 2025-06-02:
A few options come to mind, maybe even a mix of these is appropriate:

## Option 1: Encourage Database Repair 
Wrapped into this is the idea that `happi` should support outdated/bad containers, but I'd like to note that we have a helper for cleaning these up now (`happi repair`), so it's worth considering a world where the error message prompts us to repair the database, maybe even earlier in the process.

## Option 2: Be permissive without dropping data
On the other hand, it'd be really nice for old entries to continue to be usable. Is it possible to, instead of making `client["malformed"]` return `None` or raise an exception, have it return a reasonable half-entry? e.g. a partial search result that has all of the well-formed data with appropriate logged error messages and a unique class or data member to let us know that it isn't quite well-formed, or even which entries are missing/bad?

## Option 3: A separate dict-able output
Could we support something like `dict(client.iter_good_entries())`, so that it's up to the user if they care to skip the malformed entries or not?

These are my immediate thoughts. Personally I'm leaning toward something like Option 2 because it's really annoying to deal with bad data exceptions at runtime for downstream consumers and I really shouldn't care unless the specific thing I want is the data that's malformed.

tangkong, 2025-06-02:
Great thoughts.  I like option 2 as well, with a "MalformedResult" or something similar.  

I think Option 1 would be great in an ideal world, but the way our users end up on old environments we may not be able to get there.