pcdshub/happi #301: BUG: Interrupt during save operation can corrupt database

tangkong, 2023-02-22:
## Expected Behavior
Saving should always at least return the database to a readable state, but preferably complete any atomic operation before terminating.

## Current Behavior
If a save operation is interrupted, the current save operation will not be completed.  If the interruption is caught and `save()` is called again, the database can be restored. (this is the solution used in #300, but should be replaced when a more general solution is found)

The below seems to work:
```python
try:
    result.item.save()
except KeyboardInterrupt:
    result.item.save()
    break
```

## Possible Solution
Context managers that catch all exceptions and properly close out the files?  In any case we need to make the save operation safer, and more tolerant of interruption

## Steps to Reproduce (for bugs)
Start a sequence of save actions using the happi client, even without edits to an item (though last edit date will change)
Ctrl+C / KeyboardInterrupt / SigInt 
Depending on timing the database may be in various states of completeness.  

[image]

Here happi repair is run on all the devices with debug output. The keyboard interrupt occurs after the backend `.update()` method appears to resolve, but it's clear the file writing is not complete.  The diff in the json database is shown on the left. 
 Changes at the top are all valid last_edited modifications, while the bottom of the database is arbitrarily truncated.  

## Context
First uncovered when working on `happi repair` #300, but has existed for a while. 

## Your Environment
pcds-5.6.0

klauer, 2023-02-22:
I think:
* We should have the JSON backend write to a temporary file (`.db.json`, e.g.) and then move it over the database
   * This is much safer and pretty much can't lead to a truncated file
* And/or we should consider masking `KeyboardInterrupt` when `save()` happens

ZLLentz, 2023-02-22:
Building a second file and moving/copying it on top of the original is the normal way to do atomic updates safely. This affects only the json backend, correct?

tangkong, 2023-02-22:
I've never tested a different backend, but I'd (maybe too optimistically) expect a mongo backend to behave better

klauer, 2023-02-22:
It's the database's job to handle atomic updates coherently/safely - so yes, our JSON backend should be the only one affected.

KeyboardInterrupt could stop a transaction from taking place with mongo or failing to get if the insert/update succeeded.

PR #303 (2023-03-03): FIX: json backend atomic writes

## Description

Use the operating system's copy/rename/move mechanisms to implement atomic updates of the json backend.
Accordingly, the file locking has been removed, it should no longer be necessary.

Also remove some references to Travis CI that I noticed while filling in the template.

## Motivation and Context

closes #301 
Summary: if you press ctrl+c at the wrong time you can truncate your database if it uses the json backend.

## How Has This Been Tested?

I added some unit tests and the old tests still pass.
Interactively with a test db, I am able to c [... cut]
closed 2023-03-03 by ZLLentz