pcdshub/happi #325: Direct dependency for mongo backend 'bson' not listed as dev-requirement

klauer, 2023-08-21:
## Expected Behavior
Any import which is directly imported should be represented in our `requirements.txt` for package-wide requirements or `dev-requirements.txt` for "full functionality" with optional dependencies (such as here in the mongo backend)
`bson` is used in the (optional) mongo backend so it should be a `dev-requirement`

## Current Behavior
`bson` is used in the mongo backend but is not listed as a requirement anywhere. It's likely assumed to come from a dependency-of-a-dependency.

## Possible Solution

## Steps to Reproduce (for bugs)

Testing locally with a new conda environment:
```
$ conda create -n happi-3.11 python=3.11 ipython
$ conda activate happi-3.11
$ pip install -r requirements.txt -r dev-requirements.txt
$ pytest -v happi/tests/test_backends.py::test_multi_backend
```

Fails with:
```
ERROR happi/tests/test_backends.py::test_multi_backend - ModuleNotFoundError: No module named 'bson'
```

## Context
Testing Python 3.11-related issues

## Your Environment
OSX + conda

klauer, 2023-08-21:
I think this was user error (`PATH` was finding the wrong Python)
That said, it still should be in our dev-requirements in my opinion

tangkong, 2023-08-21:
I think this is actually a dependency / component of `pymongo`, in which case pymongo instructs us to explicitly not install bson from pypi or otherwise https://github.com/mongodb/mongo-python-driver/blob/master/pyproject.toml#installation  (pymongo packages/vendors a version of bson)

klauer, 2023-08-21:
Oh, huh... Good call!
I think it might be wise to have a commented-out version of "bson" in our requirements making note of that. It'd be rather easy to forget the reason

tangkong, 2023-08-21:
100% agreed, this is quite the strange edge case

klauer, 2023-08-21:
> Given that `bson` is supposed to come with pymongo I'm a little confused about why it didn't make it into your environment though...

User error 😁 
Sometimes my fancy development environment is too magical even for me...

commit d8efb9b0 (2023-09-12): DOC: explain bson dependency to avoid confusion, closes #325
closed 2023-09-12 by ZLLentz