pcdshub/happi #326: Pre-release notes script does not check if EDITOR is unset

klauer, 2023-08-22:
## Issue

This code in the pre-release notes script:
https://github.com/pcdshub/happi/blob/42aecf8504134bbc8d2f576c796c0a261df3f1f9/docs/pre-release-notes.sh#L32

Assumes `EDITOR` is set
And if `EDITOR` is unset, it tries to _run_ the newly-created `.rst` file (oops!)
It should warn, default to `vi` (totally unbiased), or do nothing/exit early

tangkong, 2023-08-22:
This is probably a issue for every prerelease notes script... I've just been copying it over every time I see it missing.

tangkong, 2024-08-27:
Really the place to change this would be in our cookiecutter.   After that we should change existing repos
https://github.com/pcdshub/cookiecutter-pcds-python/blob/master/%7B%7B%20cookiecutter.folder_name%20%7D%7D/docs/pre-release-notes.sh

PR #344 (2024-09-25): MNT: make pre-release notes script check if EDITOR is set

## Description

## Motivation and Context

closes #326 

## How Has This Been Tested?
Tested manually

## Where Has This Been Documented?

don't think it's necessary for this

## Pre-merge checklist
- [x] Code works interactively
- [x] Code contains descriptive docstrings, including context and API
- [ ] New/changed functions and methods are covered in the test suite where possible
- [x] Test suite passes locally
- [x] Test suite passes on GitHub Actions
- [x] Ran ``docs/pre-release-notes.sh`` and created a pre-release documentation pa [... cut]
closed 2024-09-25 by ZLLentz