pcdshub/happi #340: GUI / BLD: Make separate happi-qt sub-package?

tangkong, 2023-11-07:
## Expected Behavior
We might want to consider having a `happi[core]` and `happi[qt]` , once we start to accrue significant qt elements.

## Current Behavior
Qt widgets and their dependencies are grouped with the core code.  

## Possible Solution
[optional dependencies install instructions](https://setuptools.pypa.io/en/latest/userguide/dependency_management.html#optional-dependencies)

## Context
#246 

## Your Environment
pcds-5.8.0

untzag, 2023-11-15:
You might already be doing this, but please consider using qtpy abstraction layer. As a pyside2 user I often get bitten by people expecting pyqt to be installed. Whatever works best for you!

klauer, 2023-11-15:
https://github.com/pcdshub/happi/blob/63051eaa90ee34769de70d85d3115fc35c52133f/happi/qt/model.py#L5

tangkong, 2024-01-04:
Dumping some notes so I can find them whenever we decide to do this.  Adding sub-packages would be simple if we only used PIP, but conda feedstocks get a bit more confusing.

## Pip
For pip, we can simply define the extra dependencies in `pyproject.toml` and install via `pip install happi[qt]` ([link as above](https://setuptools.pypa.io/en/latest/userguide/dependency_management.html#optional-dependencies)).  This is rather clean and simple.  

## Conda
Conda does not have an analogous mechanism.  Instead a recipe can specify additional "outputs", which lead to the creation of "metapackages".  This is what `matplotlib` does with `matplotlib-base`  ([matplotlib example recipe](https://github.com/conda-forge/matplotlib-feedstock/blob/5af510232c2fa984e981e3eeb18128a6cab41b6d/recipe/meta.yaml#L69-L82))

```yaml
package:
  name: my_package

requirements:
  host: 
   - python
  run:
    - my_package-the_subpackage1
    - my_package-the_subpackage2

outputs:
  - name: my_package.the_subpackage1
    requirements:
      - some-dep
    script: some_build_script.sh
  - name: my_package.the_subpackage2
    requirements:
      - some-other-dep
    script: some_other_build_script.sh
```

This seems to create multiple `conda install`-able packages from one recipe, both sourced from the same feedstock.
[image]

It'd be great to have the same package names, but I think conda disallows brackets in package names.  So we'd have to do something like `conda install happi-qt`

ZLLentz, 2024-01-04:
I think `happi` (installs all the py files) and `happi-qt` (adds `qtpy` dependency and maybe `pyqt` in run_constrained) makes a lot of sense for the conda recipe. This is probably the pattern we should follow in general for packages that have optional widget components, i.e. there are qt elements provided but they aren't the main focus of the library.

tangkong, 2024-09-25:
I should revisit this, now that we have a working example of this in lightpath https://github.com/conda-forge/lightpath-feedstock/blob/main/recipe/meta.yaml

PR #353 (2025-02-10): BLD: split ui dependencies into subpackage
## Description
Splits happi gui dependencies into subpackages (pip) or separate outputs (conda)

## Motivation and Context
closes #347 
closes #340 

## How Has This Been Tested?
Added some ui tests, but mostly relying on CI here to assess failure

## Where Has This Been Documented?
This PR

## Pre-merge checklist
- [x] Code works interactively
- [x] Code contains descriptive docstrings, including context and API
- [x] New/changed functions and methods are covered in the test suite where possible
- [x] Test suite passes locally
- [x] Test suite passes [... cut]
closed 2025-02-10 by tangkong